IndustrialShields/arduino-Tools40 #1: class ModbusTCPSlave error

fnabarlats, 2019-11-01:
Hi, I'm trying implement the ModbusTCPSlave.h library.

When I create 'modbus' object;

```
ModbusTCPSlave  modbus(port);
```
the Arduino IDE (ver. 1.18.10) gives me the following error of compilation:

***error: 'ModbusTCPSlave' does not name a type; did you mean 'ModbusSlave'?***

 Any suggestions?
Thanks, Federico.

jordicasals, 2019-11-01:
Dear Federico,

We assume that you are using the included example. 

To use the ModbusTCPSlave library you have to compile it using an Industrial Shields M-Duino device. It is mandatory to install the Industrial Shields boards into the Arduino IDE. http://blog.industrialshields.com/en/installing-industrial-shields-equipment-to-the-arduino-ide/

Jordi

closed 2019-12-10 by IndustrialShields

charlieschick, 2020-01-08:
I also got this error with a sketch where I called:

`ModbusTCPSlave slave(port);`

There were no dependencies that required me to compile to the M-Duino board (all the code leading up to the error down below).

But just to check, I downloaded the IS boards to the IDE and tried to compile the ModbusTCPSlave example sketch with the M-Duino board selected. I still got: **_error: 'ModbusTCPSlave' does not name a type; did you mean 'ModbusSlave'?_**

Any idea where the library error is?

Also, you say:

> To use the ModbusTCPSlave library you have to compile it using an Industrial Shields M-Duino device. It is mandatory to install the Industrial Shields boards into the Arduino IDE.

Does the library not work with other Arduino IDE boards? In other words, the library only works with Industrial Shields boards? (that's not what it says in github).

Excuse my confusion.

```
// ethernet stuff
#include <SPI.h>
#include <Ethernet.h>
//Avoid pins 4,10,11,12,13 when using ethernet shield
byte mac[] = { 0xA8, 0x61, 0x0A, 0xAE, 0x29, 0x84 };
IPAddress ip(192, 168, 1, 177);
IPAddress myDns(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
int port = 502;

// for ModbusTCP
#include <ModbusTCPSlave.h>
// The default TCP port is the 502 but you can change it
ModbusTCPSlave slave(port);
```

charlieschick, 2020-01-08:
[image]

I just went though all the Tools40 libraries to see which ones also threw the same error. Only then did I realize the IDE was coloring the libraries it understood as classes and not the ones that a throwing the 'not name a type' error.

Does this help troubleshoot this issue? Alas, I am not library savvy enough to figure this out on my own.

jordicasals, 2020-01-08:
Hi,

try updating the Tools40 library. I think that the problem is related to
the Ethernet and Ethernet2 libraries.

Thank you,

charlieschick, 2020-01-08:
I had downloaded the library yesterday, but I uninstalled my Arduino IDE and reinstalled it and added a new download from your Github. Still same error.

I can get ALL your other examples to compile no problem. Only the ModbusTCP examples (all of them) are causing this problem. And they are the ones using Ethernet.h. 

Poking around, I also played with the header files a bit, and it does seem to be related to Ethernet. I get past this error if I comment out the #if defined(ethernet_h) wherever it shows up, but then other Ethernet issues pop up (EthernetClient not declared, and EthernetClient does not name a type, and, for some reason, errors in Pulses.h).

Ideas? I keep reading about the errors that pop up with #ifndef and nested #include. Might this be the issue?

DLuffy999, 2021-12-07:
I have same error message from ModbusTCPMasterReadInputRegisters example.
I use the Arduino Yun.
How can I fix the error?
I am an arduino beginer, please let me know the solution as possible as detailly.

charlieschick, 2021-12-08:
@DLuffy999 
I was using the Adafruit Metro M4 boards for client and server. I was using their Arduino Ethernet shield with the W5500 chipset, and used the Ethernet2 library for that.

Alas, I did not use the Industrial Shields libraries (as much as I wanted to). I used the Arduino Modbus library, and I had to go into one of the sub libraries to make it work. Here is what my notes say:

**Library modification**
If you download the Arduino Modbus library fresh, you will need to make a minor change to use with the Adafruit boards.
The Arduino Modbus library is based on the libmodbus library. There is a bug in the libmodbus library that needs to be deleted or commented out.
Located in the modbus-tcp.cpp file in the libmodbus source.
[path to sketchbook]\sketchbook\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp
```
#ifndef DEBUG
#define printf(...) {}
#define fprintf(...) {}
#endif
``` 
I changed it to this:
```
#ifndef DEBUG
// CS commented out for non Arduino boards
//#define printf(...) {}
//#define fprintf(...) {}
#endif
```

As I was short on time for the project, I used what worked for me (not to mention, I was still pretty new to the platform, process, and protocol). So I never went back to see if this was a similar issue with the Industrial Shields library. But seeing how you and I had problems when using a board that the libraries were not designed with, I would not be surprised there's some minor board-specific tweak that needs to be made.

Hope this helps.