IndustrialShields/arduino-Tools40 #7: WriteSingleRegister

porkyneal, 2020-09-24:
Hi, great set of tools and hardware many thanks.

I am experiencing a response issue with ModbusTCP.

If I use the Write Single Register function, the value is set correctly, but it seems the response to the client is incorrect (see below)

[image]

If I use multiple registeries, everything works fine and the response is accepted (see below)

[image]

I think the response should really be:

36 26 00 00 00 06 02 06 00 00 00 00

Which means the calculated length is 1 too long with an erronous 0x06 being sent at the end.

I can't quite figure out where the issue lies. Many Thanks.

porkyneal, 2020-09-24:
Same seems to be for the RTU Slave also.

porkyneal, 2020-09-24:
I have seemed to have solved the problem by adding "_next = data + 4;" see below

```
uint8_t ModbusSlave::writeSingleRegister(uint8_t *data) {
	if (!_holdingRegisters) {
		return IllegalFunction;
	}

	uint16_t addr = (data[0] << 8) + data[1];
	uint16_t value = (data[2] << 8) + data[3];

	if (addr > _numHoldingRegisters) {
		return IllegalDataAddress;
	}

	*(_holdingRegisters + addr) = value;

	// Keep the response equal to the request
	_next = data + 4;
	
	return UpdateMade;
}
```

porkyneal, 2020-09-24:
Also, you can write further than _numHoldingRegisters before an exception, should probably be

if (addr + 1 > _numHoldingRegisters)

jordicasals, 2020-09-28:
Hi Aaron,

thank you for using this library.

We will review this issues as soon as possible.

closed 2020-10-22 by IndustrialShields