IndustrialShields/arduino-Tools40 #14: Frage

Martin1683, 2022-01-25:
Guten Abend,

es geht um des Sketch: ModbusTCPMasterReadInputRegisters

Wie kann ich den Sketch erweitern, wenn ich aus mehreren Registern lesen möchte z.B.

master.readInputRegisters(slave, 100, 817, 1)
und 
master.readInputRegisters(slave, 30, 18, 1)

Wäre super, wenn Sie mir helfen könnten.

Die SlaveIP und der SlavePort bleiben gleich.

jordicasals, 2022-01-31:
Dear @Martin1683 
could you please write it in english?
Thank you,

Martin1683, 2022-02-18:
Hello everyone,

unfortunately I don't have time to write a great novel. I was able to solve the problems for my above question. Just look at my sketch:`

#include <Ethernet.h>
#include <ModbusTCPMaster.h>

#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd1(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

// Ethernet configuration values
uint8_t mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
uint8_t ip[] = { 192, 168, 178, 65 };
uint8_t slaveIp[] = { 192, 168, 178, 54 };
uint16_t slavePort = 502;

// Define the ModbusTCPMaster object
ModbusTCPMaster master1;
ModbusTCPMaster master4;
ModbusTCPMaster master5;

// Ethernet client object used to connect to the slave
EthernetClient slave;

uint32_t lastSentTime1 = 0UL;
uint32_t lastSentTime2 = 0UL;
uint32_t lastSentTime4 = 0UL;
uint32_t lastSentTime5 = 0UL;

int L1;
int L2;
int L3;
int ACgesamt;
int SOC;
int PVpower;
int diffACPV;

//Pins
int Pumpe = 7;
int Pumpe_led = 30;
int Heizstab1 = 5;
int Heizstab1_led = 31;
int Heizstab2 = 6;
int Heizstab2_led = 32;
int S1 = 2;
int S1_led = 33;
int S2 = 3;
int S2_led = 34;
int PWM = 9;
int Ventil = 8;

////////////////////////////////////////////////////////////////////////////////////////////////////
void setup() {

lcd1.begin(20,4);  
 
pinMode(Pumpe, OUTPUT);
pinMode(Heizstab1, OUTPUT);
pinMode(Heizstab1_led, OUTPUT);
pinMode(Heizstab2, OUTPUT);
pinMode(Heizstab2_led, OUTPUT);
pinMode(S1, INPUT_PULLUP);
pinMode(S1_led, OUTPUT);
pinMode(S2, INPUT_PULLUP);
pinMode(S2_led, OUTPUT);
pinMode(PWM, OUTPUT);
pinMode(Ventil, OUTPUT);
  
  
  
[... 239 lines cut ...]
  if (diffACPV <10) lcd1.print(" ");
  lcd1.print(diffACPV);}
  if (diffACPV < 0){
  lcd1.setCursor(0,1);  
  lcd1.print("Ueberschuss: W     0"); }
  
  lcd1.setCursor(0,2);
  lcd1.print("SOC:         %   ");
  if (SOC <100) lcd1.print(" ");
  if (SOC <10) lcd1.print(" ");
  lcd1.print(SOC);
  
  lcd1.setCursor(0,3);
  lcd1.print("PV Power:    W  ");
  if (PVpower <1000) lcd1.print(" ");
  if (PVpower <100) lcd1.print(" ");
  if (PVpower <10) lcd1.print(" ");
  lcd1.print(PVpower);
//-----------------------------------------End LCD----------------------------------------------------
 
  lastSentTime2 = millis();

}

}`

grphine, 2022-02-18:
Thanks for shoowing your solution that works.
I had the idea of multiple concurrent master objects, but I didn't think it was so good and hoped there may be some recommended way to sequentially run things.

Thanks again, though

jordicasals, 2022-02-21:
Hi,

if you want a 'blocking version' of the Modbus library you can implement something like this:

```
modbusTcpMaster.readInputRegisters(slaveAddress, inputRegisterAddr, numInputRegisters);
while (modbus.isWaitingResponse()) {
  // wait for a response
}
// detect timeout
// process response
```

closed 2022-04-20 by jordicasals