IndustrialShields/arduino-Tools40 #15: Modbus RTU slave compilation error

grphine, 2022-02-21:
This doesn't appear to be a duplicate of the existing issues.
The default sketch does not compile, with the error following. I assume that the RS485 class definition isn't treated as hardware serial any longer, likely due to industrial shields updating it. That is simply a guess on my part.

Switching out the RS485 reference for Serial, Serial1 etc. compiles successfully.

```
ModbusRTUSlave:57:32: error: no matching function for call to 'ModbusRTUSlave::ModbusRTUSlave(RS485Class&, int)'
 ModbusRTUSlave modbus(RS485, 31);
                                ^
In file included from \arduino_modified_sketch_598571\ModbusRTUSlave.ino:19:0:
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:10:3: note: candidate: ModbusRTUSlave::ModbusRTUSlave(HardwareSerial&, uint8_t)
   ModbusRTUSlave(HardwareSerial &serial, uint8_t addr);
   ^~~~~~~~~~~~~~
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:10:3: note:   no known conversion for argument 1 from 'RS485Class' to 'HardwareSerial&'
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:8:7: note: candidate: constexpr ModbusRTUSlave::ModbusRTUSlave(const ModbusRTUSlave&)
 class ModbusRTUSlave : public ModbusSlave {
       ^~~~~~~~~~~~~~
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:8:7: note:   candidate expects 1 argument, 2 provided
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:8:7: note: candidate: constexpr ModbusRTUSlave::ModbusRTUSlave(ModbusRTUSlave&&)
\Arduino\libraries\arduino-Tools40-master\src/ModbusRTUSlave.h:8:7: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "RS485.h"
 Used: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\RS485
 Not used: \Arduino\libraries\ArduinoRS485
Multiple libraries were found for "Ethernet.h"
 Used: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\Ethernet
 Not used: \Arduino\libraries\Ethernet
Using library RS485 at version 1.0 in folder: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\RS485 
Using library SoftwareSerial at version 1.0 in folder: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\SoftwareSerial 
Using library arduino-Tools40-master at version 1.0.0-beta in folder: \Arduino\libraries\arduino-Tools40-master 
Using library Ethernet at version 2.0.0 in folder: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\Ethernet 
Using library SPI at version 1.0 in folder: \AppData\Local\Arduino15\packages\industrialshields\hardware\avr\1.1.38\libraries\SPI 
exit status 1
no matching function for call to 'ModbusRTUSlave::ModbusRTUSlave(RS485Class&, int)'
```

jordicasals, 2022-02-21:
Hi @grphine 

what device are you using?

Thank you,

grphine, 2022-02-21:
Mduino 21+

jordicasals, 2022-02-21:
I've got this error when I have tried to compile for a M-Duino21. Are you compiling for a M-Duino21?
If this is the case, you should select the M-Duino21+ board into the M-Duino+ Family menu.

jordicasals, 2022-02-21:
I found this error when compiling for an M-Duino21 PLC. Could you confirm that you have selected the M-Duino21+. All of them are into the "M-Duino family" menu.

grphine, 2022-02-21:
I checked, and the IDE had reset the board to 19R.
Compiling for 21+ has solved this.

closed 2022-02-21 by grphine