IndustrialShields/arduino-Tools40 #16: Example projects do not compile

dandamron, 2022-02-22:
ModbusTCPSlave does not compile - 
Arduino\libraries\arduino-Tools40-master\src/ModbusTCPSlave.h:28:16: error: 'MODBUS_TCP_ADU_SIZE' was not declared in this scope
   uint8_t _adu[MODBUS_TCP_ADU_SIZE];
                ^~~~~~~~~~~~~~~~~~~

jordicasals, 2022-02-23:
Duplicate of #12

dandamron, 2022-02-23:
I'm sorry, saying that this is the same as #12, and closing this issue is not right.
If you look at #12, the user was using an unsupported board.  I'm compiling for an ESP PLC 19R. which IS supported.
Please re-open and answer the question rather than saying its a duplicate.

jordicasals, 2022-02-23:
Hi @dandamron 

you are right, the ESP32PLC was not in the supported devices list.

Now it should be solved.

dandamron, 2022-02-23:
I have confirmed the issues I was having are now fixed.  Thank you.

closed 2022-02-23 by dandamron