IndustrialShields/arduino-Tools40 #18: Unable to connect using modbus TCPSlave example project

dandamron, 2022-03-09:
I am using an IndustrialShields 10IOs Relay Module 013002000200

I have loaded ModbusTCPSlave, and made only 1 change - IP Address.
The TCP port is 502. (unchanged)
I can ping the device, I can see it on the network.

I cannot connect to the device using either a HMI or directly via RAW tcp open - the port is not listening.

I can see this via serial console: 

13:22:50.103 -> 172.20.251.146

In the main loop, all I see to process modbus requests is line 107, modbus.update()
Am I safe in assuming that library call also handles the ethernet handling?
in setup, I see ethernet.begin(), but nowhere else do I see anything related to ethernet..

Thanks,

ganehag, 2022-04-20:
Hi,

I took a look at the example Sketch. It seems that it's missing a few things. The Sketch should really come with some error handling as it would have helped you narrow down the issue.

The 10IOs hardware is using a Wiznet W5500 to manage its Ethernet connectivity. This chip sits on the SPI bus and requires enabling a SS (slave-select) pin for the chip to be active. On the ESP32 development board used in the 10IOs, this is pin 33.

Thus, try adding the line; `Ethernet.init(33);` right before `Ethernet.begin(mac, ip);`. It's should solve the problem as this is required for any Ethernet code to work on the 10IOs.

Gusev-Roman, 2022-04-20:
on my Mega-2560 based PLC this example works fine.

dandamron, 2022-04-20:
I have returned the product as was not able to get it working, and struggled with it for 3 weeks, blew my deadline away. Please take me off your mailing list, I will not buy your products again.

closed 2024-01-11 by JoanVC100